esp-rs/esp-idf-template #218: "failed to run custom build command for `esp-idf-sys v0.34.1`" when running "cargo run"

MatsRivel, 2024-06-21:
## Bug description
Running "cargo run" while in WSL seems fail with "failed to run custom build command for `esp-idf-sys v0.34.1`

- Would you like to work on a fix? [y]

## To Reproduce

Steps to reproduce the behavior:

1. Follow the "Prerequisites" from this repo for Linux.
2. Generate a new project from the esp-idf template
3. Run "cargo run"

## Expected behavior

## Screenshots

`matsrivel@7V5QFY3:~/code_stuff/rust_iot$ cargo run --verbose
       Fresh core v0.0.0 (/home/matsrivel/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
       ...
       Fresh embedded-svc v0.27.1
error: failed to run custom build command for `esp-idf-sys v0.34.1`

Caused by:
  process didn't exit successfully: `/home/matsrivel/code_stuff/rust_iot/target/debug/build/esp-idf-sys-f8047b6d0b14371a/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  IDF_PYTHON_ENV_PATH=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env
  ESP_IDF_VERSION=5.1
  PATH=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/cmake/3.24.0/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/ninja/1.10.2/:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmphyob6zg1idf_18845
  Current system platform: linux-amd64
  Installing tools: xtensa-esp32-elf, cmake, ninja, esp32ulp-elf
  Skipping xtensa-esp32-elf@esp-12.2.0_20230208 (already installed)
  Skipping cmake@3.24.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env
  ESP_IDF_VERSION=5.1
  PATH=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/cmake/3.24.0/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/ninja/1.10.2/:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/tmp/tmplgwnbdrbidf_18845
  cargo:rerun-if-changed=/home/matsrivel/code_stuff/rust_iot/sdkconfig.defaults
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/home/matsrivel/code_stuff/rust_iot/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fd1f65ee259a3b67/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3" IDF_TARGET="esp32" IDF_TOOLS_PATH="/home/matsrivel/code_stuff/rust_iot/.embuild/espressif" PATH="/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3/tools:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/cmake/3.24.0/bin:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/tools/ninja/1.10.2/:/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/home/matsrivel/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/matsrivel/.cargo/bin:/home/matsrivel/.vscode-server/bin/5437499feb04f7a586f677b155b039bc2b3669eb/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/lib/wsl/lib:/mnt/c/Program Files (x86)/Microsoft SDKs/Azure/CLI2/wbin:/mnt/c/Program Files/Microsoft SDKs/Azure/CLI2/wbin:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/Microsoft SQL Server/150/Tools/Binn/:/mnt/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn/:/mnt/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit/:/mnt/c/msys64:/mnt/c/Winlibs/mingw-w64-v11.0.0/mingw-w64-v11.0.0:/mnt/c/MinGW/bin:/mnt/c/Program Files/Microsoft/Azure Functions Core Tools/:/mnt/c/Program Files/Go/bin:/mnt/c/Program Files/Docker/Docker/resources/bin:/mnt/c/Users/mats.riveland/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-clang/bin:/mnt/c/Users/mats.riveland/.rustup/toolchains/esp/xtensa-esp-elf/bin:/mnt/c/Espressif/tools/idf-python/3.11.2/:/mnt/c/Espressif/tools/idf-exe/1.0.3:/mnt/c/Espressif/tools/idf-git/2.43.0/cmd/:/mnt/c/Espressif/frameworks/esp-idf-v5.2.1-2/tools:/mnt/c/Espressif:/mnt/c/Program Files (x86)/Microsoft/Edge/Application:/mnt/c/Program Files/Microsoft SDKs/Azure/CLI2/wbin:/mnt/c/Windows/system32:/mnt/c/Windows:/mnt/c/Windows/System32/Wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0/:/mnt/c/Windows/System32/OpenSSH/:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Program Files/Git/cmd:/mnt/c/Program Files/dotnet/:/mnt/c/Program Files/Microsoft SQL Server/150/Tools/Binn/:/mnt/c/Program Files/Microsoft SQL Server/Client SDK/ODBC/170/Tools/Binn/:/mnt/c/Program Files (x86)/Windows Kits/10/Windows Performance Toolkit/:/mnt/c/msys64:/mnt/c/Winlibs/mingw-w64-v11.0.0/mingw-w64-v11.0.0:/mnt/c/MinGW/bin:/mnt/c/Program Files/Microsoft/Azure Functions Core Tools/:/mnt/c/Users/mats.riveland/AppData/Local/Programs/Python/Launcher/:/mnt/c/Users/mats.riveland/.cargo/bin:/mnt/c/Users/mats.riveland/AppData/Local/Microsoft/WindowsApps:/mnt/c/Users/mats.riveland/.dotnet/tools:/mnt/c/Users/mats.riveland/Ap:/mnt/c/Program Files (x86)/Microsoft Visual Studio/2022/BuildTools/VC/Tools/MSVC/14.34.31933/bin/Hostx64/x64:/mnt/c/Users/mats.riveland/OneDrive - Bouvet Norge AS/Documents/CP210x_VCP_Windows/CP210x_VCP_Windows:/mnt/c/Users/mats.riveland/OneDrive - Bouvet Norge AS/Documents/CP210x_VCP_Windows/CP210x_VCP_Windows/x64:/mnt/c/Users/mats.riveland/AppData/Local/Microsoft/WinGet/Packages/zig.zig_Microsoft.Winget.Source_8wekyb3d8bbwe/zig-windows-x86_64-0.12.0:/mnt/c/Users/mats.riveland/bin/dfu-util-0.11-binaries/win64:/mnt/c/Users/mats.riveland/bin/dfu-util-0.11-binaries:/mnt/c/Users/mats.riveland/bin:/mnt/c/Users/mats.riveland/AppData/Local/Programs/Microsoft VS Code/bin:/snap/bin" PROJECT_DIR="/home/matsrivel/code_stuff/rust_iot" SDKCONFIG_DEFAULTS="/home/matsrivel/code_stuff/rust_iot/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fd1f65ee259a3b67/out/gen-sdkconfig.defaults;/home/matsrivel/code_stuff/rust_iot/sdkconfig.defaults" "cmake" "/home/matsrivel/code_stuff/rust_iot/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fd1f65ee259a3b67/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/home/matsrivel/code_stuff/rust_iot/.embuild/espressif/python_env/idf5.1_py3.12_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/home/matsrivel/code_stuff/rust_iot/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-fd1f65ee259a3b67/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address" "-DCMAKE_ASM_FLAGS= -mlongcalls"
[... cut ...]
    /home/matsrivel/code_stuff/rust_iot/.embuild/espressif/esp-idf/v5.1.3/tools/cmake/project.cmake:547 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /home/matsrivel/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace`

## Environment
- OS: Ubuntu 20.04
- How did you install the environment: espup 0.12.0

Vollbrecht, 2024-06-21:
did it work previously in your wsl environment and just started to make a problem with a newly generated template? Or did it never worked? 

We also changed the way the tools installation and are close to releasing this changes. Would you be so kind to test if that would make this error disappear?

You just have to add the following into your `Cargo.toml`
```
[patch.crates-io]
embuild ={ git = "https://github.com/esp-rs/embuild"}
esp-idf-svc ={ git = "https://github.com/esp-rs/esp-idf-svc"}
esp-idf-hal ={ git = "https://github.com/esp-rs/esp-idf-hal"}
esp-idf-sys ={ git = "https://github.com/esp-rs/esp-idf-sys"}
embedded-svc ={ git = "https://github.com/esp-rs/embedded-svc"}
```

than run `cargo clean` + `cargo update` and then run `cargo build`.

MatsRivel, 2024-06-24:
Hey! Thanks for the suggestion!

I changed it as you said, though I didn't have any `[path.crates-io]` section from before.
The only difference now was that is says that none of those elements were found in the graph, so it didn't make much of a difference sadly.

I realized that I likely have been going too much back and forth and could quite possibly have messed up my environment.
Therefor I went back and deleted the entire WSL and all things associated with it, gave it a fresh install, and will be continuing from there.

Once I get back to where I was I will update and tell you whether your suggestion works for a clean install :)

MatsRivel, 2024-06-24:
Yeah, it works now and builds without issue! Thanks!

closed 2024-06-24 by MatsRivel