esp-rs/esp-idf-template #221: Cargo generate git authentication attempts failed

adrien3d, 2024-07-10:
I've been through all steps of the https://docs.esp-rs.org/std-training.
I can successfully compile and flash example firmware.

But when I want to generate a new project, with this command:
`cargo generate esp-rs/esp-idf-template cargo`

I have the following error:
```
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
Error: Please check if the Git user / repository exists.

Caused by:
    all authentication attempts failed
```

Vollbrecht, 2024-07-10:
whats your `cargo generate --version` ?

does it work if you give it the full path actively with `cargo generate --git https://github.com/esp-rs/esp-idf-template cargo`

adrien3d, 2024-07-10:
I'm using `v0.21.2` version, and I have the same error with specifying the full path (just the ⚠️   Favorite warning is not there).

Vollbrecht, 2024-07-10:
yeah the warning is non-relevant to the issue. That sounds like a strange networking problem of some sorts. You seams to be quite lucky ;D ... 

0.21.2 was released just 4 hours ago. I updated to it but it still works for me. To not get you blocked a short workaround would be to just clone the template locally `git clone https://github.com/esp-rs/esp-idf-template` to where you want and than runt `cargo generate --path="<path to the clone>" cargo`

adrien3d, 2024-07-10:
Thanks, it is working with this workaround.

closed 2024-07-10 by adrien3d