esp-rs/esp-idf-template #222: Where to set the logging level at compile-time?

victorbnl, 2024-07-17:
Where is the best place to set the logging level at compile time with this project template? Running `RUST_LOG="debug" cargo run` does not print debug logs, I’m guessing because the variable is not passed to the ESP’s runtime environment.

Vollbrecht, 2024-07-17:
First you need to answer the following question: Log level of "what", as there are exist a bunch of different loglevel's for different parts. For example 2nd stage bootloader and your application code in general having different knob's. 

For most of the IDF logging you can set different sdkconfig levels depending on the level. Have a look at this [options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-log-default-level). For example to set the default log level to error only you can add `CONFIG_LOG_DEFAULT_LEVEL_ERROR=y` to your sdkconfig. 

You can have different sdkconfig files for different esp variants and profiles. For info look at our [build option chapter ](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_sdkconfig_defaults-esp_idf_sdkconfig_defaults)in esp-idf-sys and check the [official docu ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#custom-sdkconfig-defaults)on that topic.

If you want to reduce what is actually binary size you can also use the [CONFIG_LOG_MAXIMUM_LEVEL](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-log-maximum-level) sdkconfig option.

For other runtime logging log level changes you might want to you can simply initialize the EspLogger and set its target with something like this
```rust
    esp_idf_svc::log::EspLogger::initialize_default();
    let logger = esp_idf_svc::log::EspLogger;
    logger.set_target_level("*", LevelFilter::Trace).unwrap();
 ```

closed 2024-07-17 by Vollbrecht

victorbnl, 2024-07-18:
I meant the logging level of the logs I print with the `log` crate, sorry for the lack of precision. The issue with your solution is the log level is hard-coded. Is there a place in configuration files I can put the desired logging level? Or should I use conditional compilation and do something like

```
#[cfg_attr(log_level = "debug")]
logger.set_target_level("*", LevelFilter::Debug)?;
#[cfg_attr(log_level = "info")]
logger.set_target_level("*", LevelFilter::Info)?;
#[cfg_attr(log_level = "warning")]
logger.set_target_level("*", LevelFilter::Warning)?;
...
```

victorbnl, 2024-07-18:
Thank you very much, very interesting explanation! So as per your instructions I tried the following steps

- Set `CONFIG_LOG_MAXIMUM_LEVEL=5` in `sdkconfig.defaults`
- Set `CONFIG_LOG_DEFAULT_LEVEL=5` in `sdkconfig.defaults`
- Enable the features `log = { version = "0.4", features = ["max_level_trace", "release_max_level_trace"] }` in `Cargo.toml`
- Set the max level at runtime on the log module `log::set_max_level(log::LevelFilter::Trace);`

And when I say the following steps I mean I did them all then ran the projects, with all the changes done. And after that, my debug messages still don’t get logged.

ivmarkov, 2024-07-18:
There was some weird cludge w.r.t. how `CONFIG_LOG_MAXIMUM_LEVEL` is actually set. Try
`CONFIG_LOG_MAXIMUM_LEVEL_VERBOSE=y`

ivmarkov, 2024-07-18:
And also `CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y`.

Once these trigger, you can reduce from there.

Vollbrecht, 2024-07-18:
Overall your summary is correct, though the initial question was how to set it for different profiles. E.g set a different level for a release build compared to a debug build, or set a different level for lets say building against esp32c3 compared to building against a esp32 all at compile time. 

I linked it in my initial answer but just to make it super clear by explicitly stating how one can do that:
 - For a release and debug profile you just crate 2 sdkconfig files named `sdkconfig.defaults.debug` and `sdkconfig.defaults.release` 
 - Then you put in your `release` sdkconfig file, the following as an example `CONFIG_LOG_DEFAULT_LEVEL_ERROR=y` to only log errors and in your debug build `CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y` to log everything.

Nothing more is needed in this case, from a "global" app logging perspective. Also no other additions are needed inside your rust code. We automatically detect different sdkconfigs and apply them accordingly. 

So you now can run `cargo build` and `cargo build --release` and it just works..

victorbnl, 2024-07-19:
So now I have the following:

- `features = ["max_level_trace", "release_max_level_trace"]` in `Cargo.toml`
- `log::set_max_level(log::LevelFilter::Trace);` in code
- and my sdkconfig.defaults:
  ```
  CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
  CONFIG_LOG_MAX_LEVEL_VERBOSE=y
  ```
- finally I did a `cargo clean`

Still no debug logs printed.

ivmarkov, 2024-07-19:
Sorry, but I just repeated your setup, and it works for me. 

Here's my `sdkconfig.defaults`; notice that if you go with it, you need to set `CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=4096` as you are putting ESP IDF _itself_ in verbose logging mode, and apparently it needs extra stack for the idle task.

```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8000

CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=4096

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granuality for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n

CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
CONFIG_LOG_MAX_LEVEL_VERBOSE=y
```

W.r.t. Rust, `main.rs` is simply:
```rust
fn main() {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    log::debug!("Hello, world!");
}
```

... and NO other changes.

> * `features = ["max_level_trace", "release_max_level_trace"]` in `Cargo.toml`

Not necessary. This is the default anyway

> * `log::set_max_level(log::LevelFilter::Trace);` in code

Not necessary. 

> * and my sdkconfig.defaults:
>   ```
>   CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
>   CONFIG_LOG_MAX_LEVEL_VERBOSE=y
>   ```

This is the only necessary thing.

> * finally I did a `cargo clean`
> 
> Still no debug logs printed.

Are you building in release or debug? Though it should not matter...

ivmarkov, 2024-07-19:
Related: https://github.com/esp-rs/esp-idf-svc/pull/458