esp-rs/esp-idf-template #225: Could not read from remote repository when using `cargo generate`

HalfVoxel, 2024-07-22:
## Bug description

Running 

```
cargo generate https://github.com/esp-rs/esp-idf-template cargo
```

Fails with 
```
⚠️   Favorite `https://github.com/esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template
Error: Please check if the Git user / repository exists.

Caused by:
    could not read from remote repository; class=Net (12); code=Eof (-20)
```
- Would you like to work on a fix? y (but I have no idea where to even start)

I can reproduce this in multiple shells.

## To Reproduce

See above.

## Expected behavior

It starts generating the project.

## Environment

```
Ubuntu 22.04.4 LTS
cargo 1.82.0-nightly (5f6b9a922 2024-07-19)
cargo generate-generate 0.21.3
git version 2.34.1
```

## Workarounds

Cloning the project manually and using `cargo generate --path ./esp-idf-template cargo` works.
Note that the CI uses this method.

LelouBil, 2024-07-31:
Maybe this is related to this issue https://github.com/cargo-generate/cargo-generate/issues/874
I know my case is related

Vollbrecht, 2024-09-07:
yeah this is fixed in the latest `cargo-generate` version 22. Ultimately that is caused by users setting a .gitconfig that rewrites https into ssh uri's. `cargo-generate` in version 22 can work with that now.

closed 2024-09-07 by Vollbrecht