esp-rs/esp-idf-template #227: Wokwi GDB VSCode launch not working

OliverRhyme, 2024-08-02:
## Bug description

Running wokwi debugger via vscode does not work with error `Unable to start debugging. The value of miDebuggerPath is invalid`

- Would you like to work on a fix? n

## To Reproduce

Steps to reproduce the behavior:

prerequisites:

- Setup wokwi vscode extension with license

1. run `cargo generate esp-rs/esp-idf-template cargo`
2. use these options
```
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Configure advanced template options? · true
✔ 🤷   Enable STD support? · true
✔ 🤷   ESP-IDF version (master = UNSTABLE) · v5.2
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · true
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · true
✔ 🤷   Add CI files for GitHub Action? · false
```
3. open project in vscode
4. reopen in dev container
5. run `cargo build`
6. run launch in vscode the "Wokwi GDB"

## Expected behavior
Should be able to run wokwi gdb

## Screenshots
N/A

## Environment

- OS: MacOS Sonoma
- How did you install the environment: dev container

## Additional context

Manu343726, 2025-02-10:
@SergioGasquez Hi,

I'm working on a `no_std` project using [esp-generate](https://github.com/esp-rs/esp-generate) targeting esp32, but since the template structure is practically the same I stumbled upon this same issue. I hope you can help me figure this out.

After seeing your comment I patched the .devcontainer Dockerfile to download and install xtensa-esp-elf-gdb alongside the esp toolchain already in the container/Dockerfile. After that, only the `no-python` variant of gdb seems to work (I have seen issues talking about gdb not picking the python libs included in the toolchain in the esp-binutils project so I'll ignore this for now).

So, if I launch the Wikwi simulator followed by pressing F5 to run the debugger it looks like gdb is able to attach to the simulator, but I get the error "Remote 'g' packet reply is too long. Expected X bytes, got Y" which [the Wokwi docs](https://docs.wokwi.com/vscode/debugging#troubleshooting) state it's usually related to using a gdb for the wrong architecture.

While setting up the Dockerfile to download xtensa-esp-elf-gdb I found a bit confusing that the triple used for naming the gdb release packages did not include the 32 bit suffix while the releases of esp toolchain are named xtensa-esp32-elf. Maybe I am picking the wrong debugger after all?

Thank you.

SergioGasquez, 2025-02-10:
> Maybe I am picking the wrong debugger after all?

You'll need to download: https://github.com/espressif/binutils-gdb/releases/tag/esp-gdb-v15.2_20241112 for your target, unzip it and point to `$DOWNLOAD_FOLDER/bin/xtensa-esp32-elf-gdb`

Manu343726, 2025-02-10:
Yeah, I was asking because as I said `esp` project releases for esp32 target are named `xtensa-esp32-elf` but I don't see that in `binutils-gdb` releases, only `xtensa-esp-elf`. Is the `xtensa-esp-elf` gdb package the correct one for the esp32 target? (That is the one I picked).

SergioGasquez, 2025-02-12:
>  Is the xtensa-esp-elf gdb package the correct one for the esp32 target?

I think the correct one is `xtensa-esp32-elf-gdb`, but tbh I havent played with Wokwi debugging in a while and I might be wrong