esp-rs/esp-idf-template #241: Dev container *Hellow World* not building

Pandananana, 2024-10-19:
## Bug description

I can't build the Hello World template in the devcontainer using Docker. I get this error, I don't understand which submodules i need to update. 

```
esp@bdaee5d6e808:~/project_nexus$ cargo build
   Compiling compiler_builtins v0.1.109
   Compiling core v0.0.0 (/home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.155
   Compiling proc-macro2 v1.0.88
   Compiling unicode-ident v1.0.13
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/esp/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.161
   Compiling serde v1.0.210
   Compiling bitflags v2.6.0
   Compiling regex-syntax v0.8.5
   Compiling crossbeam-utils v0.8.20
   Compiling cfg_aliases v0.2.1
   Compiling rustix v0.38.37
   Compiling aho-corasick v1.1.3
   Compiling quote v1.0.37
   Compiling glob v0.3.1
   Compiling syn v2.0.79
   Compiling nix v0.29.0
   Compiling clang-sys v1.8.1
   Compiling prettyplease v0.2.22
   Compiling linux-raw-sys v0.4.14
   Compiling syn v1.0.109
   Compiling log v0.4.22
   Compiling rustversion v1.0.18
   Compiling crossbeam-epoch v0.9.18
   Compiling bstr v1.10.0
   Compiling regex-automata v0.4.8
   Compiling serde_json v1.0.131
   Compiling shlex v1.3.0
   Compiling thiserror v1.0.64
   Compiling same-file v1.0.6
   Compiling either v1.13.0
   Compiling minimal-lexical v0.2.1
   Compiling walkdir v2.5.0
   Compiling nom v7.1.3
   Compiling crossbeam-deque v0.8.5
   Compiling globset v0.4.15
   Compiling libloading v0.8.5
   Compiling cvt v0.1.2
   Compiling home v0.5.9
   Compiling ryu v1.0.18
   Compiling anyhow v1.0.90
   Compiling heck v0.4.1
   Compiling once_cell v1.20.2
   Compiling itoa v1.0.11
   Compiling bindgen v0.69.5
   Compiling fs_at v0.2.1
   Compiling cexpr v0.6.0
   Compiling which v4.4.2
   Compiling ignore v0.4.23
   Compiling regex v1.11.0
   Compiling itertools v0.12.1
   Compiling cc v1.1.31
   Compiling lazy_static v1.5.0
   Compiling rustc-hash v1.1.0
   Compiling fastrand v2.1.1
   Compiling lazycell v1.3.0
   Compiling normpath v1.3.0
   Compiling bitflags v1.3.2
   Compiling globwalk v0.8.1
   Compiling serde_derive v1.0.210
[... 458 lines cut ...]

- Would you like to work on a fix? Yes

## To Reproduce

Steps to reproduce the behavior:

1. Run cargo generate esp-rs/esp-idf-template cargo
2. Use advanced template options
3. Select use dev container
4. Open in VS Code
5. Run `cargo build`

## Expected behavior

Expected it to run without errors

## Screenshots

## Environment

- OS: Nixos
- How did you install the environment: devcontainer option in the template

## Additional context

Pandananana, 2024-10-20:
After changing from ESP-iDF version 5.2 to 5.1, it worked:

This is the full config:
✔ 🤷   Which MCU to target? · esp32
✔ 🤷   Configure advanced template options? · true
✔ 🤷   Enable STD support? · true
✔ 🤷   ESP-IDF version (master = UNSTABLE) · v5.1
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · true
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · true
✔ 🤷   Add CI files for GitHub Action? · true

Note that I also tested version 5.2 (which doesn't work in the esp-idf-template) on the esp-template (no_std), and there it worked.

SergioGasquez, 2024-10-21:
Thanks for posting your working setup!

closed 2024-10-21 by SergioGasquez