esp-rs/esp-idf-template #242: Version mismatch between tutorial-recommended ESP-IDF and template dependencies

wuertele, 2024-10-22:
## Issue Description
Following the official ESP-RS documentation to set up a new project results in build failures due to version incompatibilities between ESP-IDF and esp-idf-svc.

The [official documentation](https://docs.esp-rs.org/book/introduction.html) guides users to install ESP-IDF 5.0, but the template generates a project using esp-idf-svc 0.49, which is incompatible with ESP-IDF 5.0+. This creates a confusing experience for new users who follow the documentation exactly but end up with unbuildable projects.

## Steps to Reproduce
1. Follow the installation steps at https://docs.esp-rs.org/book/installation/rust.html:
   ```bash
   curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
   cargo install espup
   espup install
   cargo install cargo-generate
   cargo install ldproxy
   ```

2. Install ESP-IDF 5.0 as per documentation:
   ```bash
   mkdir -p ~/esp
   cd ~/esp
   git clone --recursive https://github.com/espressif/esp-idf.git
   cd esp-idf
   ./install.sh
   source ./export.sh
   ```

3. Generate a new project:
   ```bash
   cargo generate esp-rs/esp-idf-template
   # Select 'cargo' template and 'esp32' target
   ```

4. Try to build:
   ```bash
   cd <generated-project>
   cargo build
   ```

## Error Output
```
Compiling embedded-svc v0.28.0
error[E0063]: missing field `rssi_5g_adjustment` in initializer of `esp_idf_hal::sys::wifi_scan_threshold_t`
   --> /Users/dave/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/wifi.rs:189:24
    |
189 |             threshold: wifi_scan_threshold_t {
    |                        ^^^^^^^^^^^^^^^^^^^^^ missing `rssi_5g_adjustment`

error[E0063]: missing field `duty_cycle_pos` in initializer of `esp_idf_hal::sys::sdspi_device_config_t`
  --> /Users/dave/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.49.1/src/sd/spi.rs:41:29
   |
41 |         let configuration = sdspi_device_config_t {
   |                             ^^^^^^^^^^^^^^^^^^^^^ missing `duty_cycle_pos`
```

## Environment
- espup version: 0.12.2
- ESP-IDF version: 5.0
- Generated project's esp-idf-svc version: 0.49
- Operating System: macOS
- Rust toolchain: esp (from rust-toolchain.toml)

## Additional Context
The current template generates a Cargo.toml that includes:
```toml
[dependencies]
esp-idf-svc = { version = "0.49", default-features = false }
```

This version of esp-idf-svc appears to be incompatible with ESP-IDF 5.0 structures, leading to compilation errors due to missing fields in struct definitions.

## Suggested Solutions
1. Update the template to use a version of esp-idf-svc that's compatible with ESP-IDF 5.0
2. Add version checking/warnings during project generation
3. Update the documentation to specify compatible versions of all components
4. Consider maintaining different template branches for different ESP-IDF major versions

## Documentation References
- [Installation Guide](https://docs.esp-rs.org/book/installation/rust.html)
- [Project Generation Guide](https://docs.esp-rs.org/book/writing-your-own-application/generate-project/index.html)

wuertele, 2024-10-22:
I am doing this in the first place because my naive reading of the esp-rs book seemed to suggest it.  I didn't know anything about the esp-idf repo, and I wouldn't have tried to clone it except for trying to follow the esp-rs book.  But now that I look through the esp-rs book, I can't find the page that said I should install the esp-idf repo.  So this is almost definitely a case of misleading myself about the required dependencies. 
I removed the esp-idf repo and started a new shell without the esp-idf environment, and now when I generate using either the esp-rs/esp-template or the esp-rs/esp-idf-template, cargo build succeeds.
Thanks for the hint, this issue can be closed.

closed 2024-10-22 by wuertele

ivmarkov, 2024-10-22:
> I am doing this in the first place because my naive reading of the esp-rs book seemed to suggest it. I didn't know anything about the esp-idf repo, and I wouldn't have tried to clone it except for [...]

Hmmm, as I said, since you are not the first one to stumble on this (we just had another case a couple of weeks ago) and some others in the past, it _must_ be something in the docu.

Well, the plan is anyway to not have any installation instructions in the Book itself, but just a link to the `esp-idf-template` README. And there I'll try to make it crystal clear in the next days/weeks, that you don't have to deal with ESP-IDF unless you want to (i.e. vendored use cases etc.).