esp-rs/esp-idf-template #244: Dev container fails to start

perotom, 2024-11-01:
## Bug description

VS Code throws an error when opening a newly created project with a dev container.

```
[2024-11-01T10:03:36.690Z] Dev Containers 0.388.0 in VS Code 1.95.1 (65edc4939843c90c34d61f4ce11704f09d3e5cb6).
[2024-11-01T10:03:36.690Z] Start: Resolving Remote
[2024-11-01T10:03:36.702Z] Setting up container for folder or workspace: /Users/username/Documents/Rust/test-software
[2024-11-01T10:03:36.702Z] Context: desktop-linux
[2024-11-01T10:03:36.703Z] Start: Check Docker is running
[2024-11-01T10:03:36.704Z] Start: Run: docker version
[2024-11-01T10:03:36.725Z] Client:
 Version:           27.2.0
 API version:
[2024-11-01T10:03:36.725Z]        1.47
 Go version:        go1.21.13
 Git commit:        3ab4256
 Built:             Tue Aug 27 14:14:45 2024
 OS/Arch:           darwin/arm64
 Context:           desktop-linux

Server: Docker Desktop 4.34.3 (170107)
 Engine:
  Version:          27.2.0
  API version:      1.47 (minimum version 1.24)
  Go version:       go1.21.13
  Git commit:       3ab5c7d
  Built:            Tue Aug 27 14:15:41 2024
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.7.20
  GitCommit:        8fc6bcff51318944179630522a095cc9dbf9f353
 runc:
  Version:          1.1.13
  GitCommit:        v1.1.13-0-g58aa920
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
[2024-11-01T10:03:36.727Z] Stop (23 ms): Run: docker version
[2024-11-01T10:03:36.727Z] Stop (24 ms): Check Docker is running
[2024-11-01T10:03:36.728Z] Start: Run: docker volume ls -q
[2024-11-01T10:03:36.740Z] Stop (12 ms): Run: docker volume ls -q
[2024-11-01T10:03:36.740Z] Start: Run: docker ps -q -a --filter label=vsch.local.folder=/Users/username/Documents/Rust/test-software --filter label=vsch.quality=stable
[2024-11-01T10:03:36.752Z] Stop (12 ms): Run: docker ps -q -a --filter label=vsch.local.folder=/Users/username/Documents/Rust/test-software --filter label=vsch.quality=stable
[2024-11-01T10:03:36.752Z] Start: Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software --filter label=devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json
[2024-11-01T10:03:36.764Z] Stop (12 ms): Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software --filter label=devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json
[2024-11-01T10:03:36.764Z] Start: Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software
[2024-11-01T10:03:36.775Z] Stop (11 ms): Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software
[2024-11-01T10:03:36.775Z] Start: Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software
[2024-11-01T10:03:36.787Z] Stop (12 ms): Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software
[2024-11-01T10:03:36.788Z] Running Dev Containers CLI:   up --user-data-folder /Users/username/Library/Application Support/Code/User/globalStorage/ms-vscode-remote.remote-containers/data --container-session-data-folder /tmp/devcontainers-946948ea-b831-4daa-8d34-59f2b03337c61730455416273 --workspace-folder /Users/username/Documents/Rust/test-software --workspace-mount-consistency cached --gpu-availability detect --id-label devcontainer.local_folder=/Users/username/Documents/Rust/test-software --id-label devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --log-level debug --log-format json --config /Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vscode,target=/vscode,external=true --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root --include-configuration --include-merged-configuration
[2024-11-01T10:03:36.801Z] ProxyResolver#loadSystemCertificates count
[2024-11-01T10:03:36.802Z] ProxyResolver#loadSystemCertificates count filtered
[2024-11-01T10:03:36.802Z] Start: Run: /Applications/Visual Studio Code.app/Contents/Frameworks/Code Helper (Plugin).app/Contents/MacOS/Code Helper (Plugin) /Users/username/.vscode/extensions/ms-vscode-remote.remote-containers-0.388.0/dist/spec-node/devContainersSpecCLI.js up --user-data-folder /Users/username/Library/Application Support/Code/User/globalStorage/ms-vscode-remote.remote-containers/data --container-session-data-folder /tmp/devcontainers-946948ea-b831-4daa-8d34-59f2b03337c61730455416273 --workspace-folder /Users/username/Documents/Rust/test-software --workspace-mount-consistency cached --gpu-availability detect --id-label devcontainer.local_folder=/Users/username/Documents/Rust/test-software --id-label devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --log-level debug --log-format json --config /Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vscode,target=/vscode,external=true --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root --include-configuration --include-merged-configuration
[2024-11-01T10:03:36.931Z] @devcontainers/cli 0.71.0. Node.js v20.18.0. darwin 23.5.0 arm64.
[2024-11-01T10:03:36.931Z] Start: Run: docker buildx version
[2024-11-01T10:03:36.974Z] Stop (43 ms): Run: docker buildx version
[2024-11-01T10:03:36.974Z] github.com/docker/buildx v0.16.2-desktop.1 081c21b9e461293ae243a1ff813a680a4f5f8fb9
[2024-11-01T10:03:36.975Z] 
[2024-11-01T10:03:36.975Z] Start: Run: docker -v
[2024-11-01T10:03:36.989Z] Stop (14 ms): Run: docker -v
[2024-11-01T10:03:36.989Z] Start: Resolving Remote
[2024-11-01T10:03:36.996Z] Loading 11 extra certificates from /var/folders/w1/0j27_v7d3ls0jpf1xj41kgh80000gn/T/vsch/certificates-6a04976dd082fa62c2e87505bf3a2bf499e1642b59a06fae5edbf3115ecdf43c.pem.
[2024-11-01T10:03:37.069Z] Start: Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software --filter label=devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json
[2024-11-01T10:03:37.088Z] Stop (19 ms): Run: docker ps -q -a --filter label=devcontainer.local_folder=/Users/username/Documents/Rust/test-software --filter label=devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json
[2024-11-01T10:03:37.090Z] Start: Run: docker inspect --type image debian:bookworm-slim
[2024-11-01T10:03:37.106Z] Stop (16 ms): Run: docker inspect --type image debian:bookworm-slim
[2024-11-01T10:03:37.470Z] Start: Run: docker-credential-desktop get
[2024-11-01T10:03:37.510Z] Stop (40 ms): Run: docker-credential-desktop get
[... 81 lines cut ...]
[2024-11-01T10:03:40.253Z]     at async wrA (/Users/username/.vscode/extensions/ms-vscode-remote.remote-containers-0.388.0/dist/spec-node/devContainersSpecCLI.js:663:203)
[2024-11-01T10:03:40.253Z]     at async DrA (/Users/username/.vscode/extensions/ms-vscode-remote.remote-containers-0.388.0/dist/spec-node/devContainersSpecCLI.js:662:14830)
[2024-11-01T10:03:40.253Z]     at async /Users/username/.vscode/extensions/ms-vscode-remote.remote-containers-0.388.0/dist/spec-node/devContainersSpecCLI.js:482:1190
[2024-11-01T10:03:40.255Z] Stop (3453 ms): Run: /Applications/Visual Studio Code.app/Contents/Frameworks/Code Helper (Plugin).app/Contents/MacOS/Code Helper (Plugin) /Users/username/.vscode/extensions/ms-vscode-remote.remote-containers-0.388.0/dist/spec-node/devContainersSpecCLI.js up --user-data-folder /Users/username/Library/Application Support/Code/User/globalStorage/ms-vscode-remote.remote-containers/data --container-session-data-folder /tmp/devcontainers-946948ea-b831-4daa-8d34-59f2b03337c61730455416273 --workspace-folder /Users/username/Documents/Rust/test-software --workspace-mount-consistency cached --gpu-availability detect --id-label devcontainer.local_folder=/Users/username/Documents/Rust/test-software --id-label devcontainer.config_file=/Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --log-level debug --log-format json --config /Users/username/Documents/Rust/test-software/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type [... cut ...] true --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root --include-configuration --include-merged-configuration
[2024-11-01T10:03:40.258Z] Exit code 1
```

## To Reproduce

Steps to reproduce the behavior:

1. Run `cargo generate esp-rs/esp-idf-template cargo`
2. MCU Target: esp32s3
3. Enable STD support: True
4. ESP-IDF version: v.5.2
5.  Dev Containers: True
6. Wokwi: True
7. CI Files: False
8. Open in VS Code and choose "

## Environment

- OS: MacOS 14.5 (23F79)

SergioGasquez, 2024-11-05:
Seems to be the same issue as  https://github.com/docker/docs/issues/4709

perotom, 2024-11-07:
I don't think so. It looks like https://github.com/serialport/serialport-rs/issues/153 brings this issue on MacOS. It tries to mount the folder `/run/udev` to detect serial ports but this folder doesn't exist on MacOS. The devcontainer.json looks like:

```
{
  "name": "test-project",
  // Select between image and build propieties to pull or build the image.
  // "image": "docker.io/espressif/idf-rust:esp32s3_latest",
  "build": {
    "dockerfile": "Dockerfile",
    "args": {
      "CONTAINER_USER": "esp",
      "CONTAINER_GROUP": "esp",
      "ESP_BOARD": "esp32s3"
    }
  },
  // https://github.com/serialport/serialport-rs/issues/153
  "runArgs": [
    "--mount",
    "type=bind,source=/run/udev,target=/run/udev,readonly"
  ],
  "customizations": {
    "vscode": {
      "settings": {
        "editor.formatOnPaste": true,
        "editor.formatOnSave": true,
        "editor.formatOnSaveMode": "file",
        "editor.formatOnType": true,
        "lldb.executable": "/usr/bin/lldb",
        "files.watcherExclude": {
          "**/target/**": true
        },
        "rust-analyzer.checkOnSave.command": "clippy",
        "rust-analyzer.checkOnSave.allTargets": false,
        "[rust]": {
          "editor.defaultFormatter": "rust-lang.rust-analyzer"
        }
      },
      "extensions": [
        "rust-lang.rust-analyzer",
        "tamasfe.even-better-toml",
        "serayuzgur.crates",
        "mutantdino.resourcemonitor",
        "yzhang.markdown-all-in-one",
        "ms-vscode.cpptools",
        "actboy168.tasks",
        "Wokwi.wokwi-vscode"
      ]
    }
  },
  "forwardPorts": [
    3333,
    8000
  ],
  "workspaceMount": "source=${localWorkspaceFolder},target=/home/esp/test-project,type=bind,consistency=cached",
  "workspaceFolder": "/home/esp/test-project"
}
```

joebnb, 2024-12-06:
> docker/docs#4709

i have same issue, is their anyway workaround ?

joebnb, 2024-12-06:
> I don't think so. It looks like [serialport/serialport-rs#153](https://github.com/serialport/serialport-rs/issues/153) brings this issue on MacOS. It tries to mount the folder `/run/udev` to detect serial ports but this folder doesn't exist on MacOS. The devcontainer.json looks like:
> 
> ```
> {
>   "name": "test-project",
>   // Select between image and build propieties to pull or build the image.
>   // "image": "docker.io/espressif/idf-rust:esp32s3_latest",
>   "build": {
>     "dockerfile": "Dockerfile",
>     "args": {
>       "CONTAINER_USER": "esp",
>       "CONTAINER_GROUP": "esp",
>       "ESP_BOARD": "esp32s3"
>     }
>   },
>   // https://github.com/serialport/serialport-rs/issues/153
>   "runArgs": [
>     "--mount",
>     "type=bind,source=/run/udev,target=/run/udev,readonly"
>   ],
>   "customizations": {
>     "vscode": {
>       "settings": {
>         "editor.formatOnPaste": true,
>         "editor.formatOnSave": true,
>         "editor.formatOnSaveMode": "file",
>         "editor.formatOnType": true,
>         "lldb.executable": "/usr/bin/lldb",
>         "files.watcherExclude": {
>           "**/target/**": true
>         },
>         "rust-analyzer.checkOnSave.command": "clippy",
>         "rust-analyzer.checkOnSave.allTargets": false,
>         "[rust]": {
>           "editor.defaultFormatter": "rust-lang.rust-analyzer"
>         }
>       },
>       "extensions": [
>         "rust-lang.rust-analyzer",
>         "tamasfe.even-better-toml",
>         "serayuzgur.crates",
>         "mutantdino.resourcemonitor",
>         "yzhang.markdown-all-in-one",
>         "ms-vscode.cpptools",
>         "actboy168.tasks",
>         "Wokwi.wokwi-vscode"
>       ]
>     }
>   },
>   "forwardPorts": [
>     3333,
>     8000
>   ],
>   "workspaceMount": "source=${localWorkspaceFolder},target=/home/esp/test-project,type=bind,consistency=cached",
>   "workspaceFolder": "/home/esp/test-project"
> }
> ```

i have same issue, is their anyway workaround ?

joebnb, 2024-12-06:
https://github.com/esp-rs/std-training/blob/c7325fa5dc2d42f01fa13e5905394b85df023911/.devcontainer/devcontainer.json#L11

remove  runArgs, in official training it was already removed

joebnb, 2024-12-06:
remove runArgs will have problem on macos,

```
esp@e3a5ad01216e:~/esp32s3$ cargo run
    Blocking waiting for file lock on build directory
   Compiling esp-idf-sys v0.35.0
   Compiling lllix-esp32s3 v0.1.0 (/home/esp/lllix_esp32s3)
   Compiling esp-idf-hal v0.44.1
   Compiling esp-idf-svc v0.49.1
    Finished `dev` profile [optimized + debuginfo] target(s) in 1m 24s
     Running `espflash flash --monitor target/xtensa-esp32s3-espidf/debug/lllix-esp32s3`
Error: espflash::no_serial

  × No serial ports could be detected
  help: Make sure you have connected a device to the host system. If the device is connected but not listed, try using the `--list-all-ports` flag.
```

j-g00da, 2025-03-07:
> ```
>   × No serial ports could be detected
>   help: Make sure you have connected a device to the host system. If the device is connected but not listed, try using the `--list-all-ports` flag.
> ```

Docker on mac os does not support usb passthrough: https://github.com/docker/roadmap/issues/511 https://github.com/docker/for-mac/issues/5263
There are some weird workarounds (like running docker in linux VM and forwarding usb from mac to linux and then from linux to docker...)
I think the best approach is to just use [web-flash](https://github.com/esp-rs/esp-web-flash-server) that is already installed in devcontainer and just works (tested it on my mac).
I can write some additional documentation for that because it's far from obvious. 

Also we should make udev bind conditional (only when udev exists - so we can access host usb on linux _and_ devcontainer works on mac.) This would resolve the initial issue.