esp-rs/esp-idf-template #246: esp8266 compiled, but not works.

NsdHSO, 2024-11-20:
I want to be able toggle the default led from esp8266. For the moment I want to run a loop in the esp, and to see the logs.

For the moment I builded this program
```
fn main() {
    esp_idf_sys::link_patches();
    // Loop to toggle the LED 20 times
    for i in 0..20 {
        println!("{} Toggle", i);
    }
}

```
cargo build

all works good.

but when I want to flesh, Flashing has completed, but I got something like.

```
espflash flash target/xtensa-esp32-espidf/debug/blink --baud=460800 --monitor
[2024-11-20T09:15:55Z INFO ] 🚀 A new version of espflash is available: v3.2.0
[2024-11-20T09:15:55Z INFO ] Detected 4 serial ports
[2024-11-20T09:15:55Z INFO ] Ports which match a known common dev board are highlighted
[2024-11-20T09:15:55Z INFO ] Please select a port
[2024-11-20T09:15:56Z INFO ] Serial port: '/dev/cu.usbserial-2140'
[2024-11-20T09:15:56Z INFO ] Connecting...
[2024-11-20T09:15:57Z INFO ] Using flash stub
Chip type:         esp8266
Crystal frequency: 26MHz
Flash size:        4MB
Features:          WiFi
MAC address:       8c:ce:4e:c1:d1:81
App size:          474,048 bytes
[00:00:49] [========================================]     463/463     0x0  [2024-11-20T09:16:47Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit
``` 

**### 
_��{ll��|�l�<

            �
             d�c|����s�c�
                         b��gn�do'���
                                     #x��lrdrd8�g�
                                                  �

                                                   c
                                                    o�<l�
                                                         #��o'�d��$`�'
                                                                      d`'{���g
   �lx�o�
         �

          ��b
             n�<
                �;�d��c��'o�
                            $ �'od�d`g{���g
                                           d``�'
                                                ��c�ol�
                                                       ��ng�
                                                            �dp�'�
                                                                  ;���x�

                                                                        cg�|
 ��c��'o�$�l`�o'
                �dg{���g
                        ���`p�n�
                                ;���
                                    �
                                     cg�|Č�_**
                                           #��o'��d`�ngl`os�ۓo
                                                              ۓ`8�g�
                                                                    {���l�
                                                                          co�<l�

     �
      #��o'�l`�o'l`gs�ۓo�l`8�o�
                               ���
                                  b
                                   '�|�d�Ǐ
                                          �c��'o�
                                                 $`�'l�l or�ۓo��l{��g
                                                                     �
                                                                      �$r��'l�|��`�s��o�#�gc8�

```

NsdHSO, 2024-11-20:
If anyone has encountered this problem before and could help me, I would be grateful.

ivmarkov, 2024-11-20:
> If anyone has encountered this problem before and could help me, I would be grateful.

esp8266 will never be supported by the `esp-idf-*` crates.
Initially, we implemented "some" support, in the hope/anticipation that ESP-IDf will be ported to esp8266, but this would never happen.

We need to remove any reference to `esp8266` from `esp-idf-sys`.

ivmarkov, 2024-11-20:
The closest supported chip to esp8266 is esp32c3.
You can use any `esp32*` chip but it seems esp32c3 specifically is even pin-compatible with esp8266.

closed 2024-11-20 by ivmarkov

NsdHSO, 2024-11-20:
> esp8266 will never be supported by the `esp-idf-*` crates. Initially, we implemented "some" support, in the hope/anticipation that ESP-IDF will be ported to esp8266, but this would never happen.
>
> We [...]

Please can you explain to me why?

ivmarkov, 2024-11-20:
> Please can you explain to me why?

Because it does not work, will never work, and there is nobody who is willing to support esp8266. It is really time to move on to a newer chip.