esp-rs/esp-idf-template #247: Completely broken on Windows

LeagueRaINi, 2024-11-20:
## Bug description

- Would you like to work on a fix? [y/n]

no

## To Reproduce

Steps to reproduce the behavior:

1. cargo generate esp-rs/esp-idf-template cargo
2. Which MCU to target? · esp32c3
3. Configure advanced template options? · true
4. ESP-IDF version (master = UNSTABLE) · v5.2
5. Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · true
6. Configure project to support Wokwi simulation with Wokwi VS Code extension? · false
7. Add CI files for GitHub Action? · false

## Expected behavior
A working template with the options i choose

## Screenshots

## Environment

- OS: Windows 11 (Version 10.0.22631 Build 22631)

## Additional context

- Fails mounting `/run/udev` which does not seem to be a thing in wsl which in my case i just replaced with `/dev/bus/usb` in the devcontainers `runArgs`
- Still generated the wokwi scripts and installed the extension despite saying no to it, the scripts dont work either:
```
/bin/bash: line 1: scripts/build.sh: cannot execute: required file not found
```
- Rust analyzer doesnt work either cause its trying to use (presumibly) the container's architecture:
```
2024-11-20T19:28:23.310213273Z ERROR Flycheck failed to run the following command: CommandHandle { program: "/home/esp/.cargo/bin/cargo", arguments: ["clippy", "--workspace", "--message-format=json-diagnostic-rendered-ansi", "--manifest-path", "/home/esp/owie_c3/Cargo.toml", "--keep-going"], current_dir: Some("/home/esp/owie_c3") }, error=Cargo watcher failed, the command produced no valid metadata (exit code: ExitStatus(unix_wait_status(256))):
error: 'cargo-clippy' is not installed for the custom toolchain '/home/esp/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu'.
note: this is a custom toolchain, which cannot use `rustup component add`
help: if you built this toolchain from source, and used `rustup toolchain link`, then you may be able to build the component with `x.py`
```
- Building with `cargo build --target riscv32imc-esp-espidf` fails:
```
  Using esp-idf v5.2.2 at '/home/esp/owie_c3/.embuild/espressif/esp-idf/v5.2.2'
  thread 'main' panicked at /home/esp/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embuild-0.32.0/src/espidf.rs:153:33:
  Failed to run command: "/home/esp/owie_c3/.embuild/espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc" "--version"
```

I've worked with rust on my pico's before and expected this to be more or less painless as well but so far its a complete disaster and very frustrating, any help or proper instructions on how to get this running in wsl2 would be appreciated.

ivmarkov, 2024-11-22:
If you are willing to give it a try without containers, I can help along the way.

Otherwise, perhaps @SergioGasquez can help in here?

j-g00da, 2025-03-09:
@LeagueRaINi could you try to delete `.embuild` and `target` directories and try again? I think you are experiencing the same issue as I did and this is the current workaround.

I'm working on a fix.

ivmarkov, 2025-08-13:
Also here:
1. English please
2. You are likely missing GIT and Python3 on your windows machine