esp-rs/esp-idf-template #250: error: Toolchain esp in path_project\rust-toolchain.toml is custom and not installed

zhenchuanpeng, 2024-12-04:
2024-12-04T02:44:30.1924937+08:00 ERROR FetchWorkspaceError: rust-analyzer failed to load workspace: Failed to load the project at path_project\Cargo.toml: Failed to query rust toolchain version at path_project, is your toolchain setup correctly?: "C:\\Users\\Administrator\\.cargo\\bin\\cargo.exe" "--version" failed, exit code: 1
stderr:
error: Toolchain esp in path_project\rust-toolchain.toml is custom and not installed

[toolchain]
channel = "esp"
"I've encountered an issue where my project runs fine when I switch to 'stable' but errors out when I use 'esp'. All dependencies are installed, including espup, esp-idf v5.3.1, espflash, and ldproxy. Could you please help me troubleshoot this issue?"

zhenchuanpeng, 2024-12-04:
esp32s3

ivmarkov, 2024-12-07:
> stderr:
error: Toolchain esp in path_project\rust-toolchain.toml is custom and not installed

`espup install`

closed 2024-12-07 by ivmarkov