esp-rs/esp-idf-template #252: esp-idf-svc Not compiling from template project

piemanau, 2024-12-23:
## Bug description

Not compiling `esp-idf-svc` when using the generated template from here

- Would you like to work on a fix? [y/n]

## To Reproduce

Steps to reproduce the behavior:

1. Install prerequisates
2. Download template
3. Attempt to build

## Expected behavior

Build the templates example code

## Screenshots

## Environment

- OS: macOs 15
- How did you install the environment: [e.g. espup 0.3.2]

## Additional context

```
Compiling esp-idf-svc v0.49.1
error[E0308]: mismatched types
   --> /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.49.1/src/tls.rs:214:36
    |
214 |                 rcfg.alpn_protos = bufs.alpn_protos.as_mut_ptr();
    |                 ----------------   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*mut *const u8`, found `*mut *const i8`
    |                 |
    |                 expected due to the type of this binding
    |
    = note: expected raw pointer `*mut *const u8`
               found raw pointer `*mut *const i8`

error[E0308]: mismatched types
     --> /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.49.1/src/tls.rs:653:25
      |
652   |                     sys::esp_tls_conn_new_async(
      |                     --------------------------- arguments to this function are incorrect
653   |                         host.as_bytes().as_ptr() as *const i8,
      |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*const u8`, found `*const i8`
      |
      = note: expected raw pointer `*const u8`
                 found raw pointer `*const i8`
note: function defined here
     --> /Users/user/Documents/Projects/test-esp32c3-app/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-eeec4fa03cb86953/out/bindings.rs:40961:12
      |
40961 |     pub fn esp_tls_conn_new_async(
      |            ^^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
     --> /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.49.1/src/tls.rs:661:25
      |
660   |                     sys::esp_tls_conn_new_sync(
      |                     -------------------------- arguments to this function are incorrect
661   |                         host.as_bytes().as_ptr() as *const i8,
      |                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ expected `*const u8`, found `*const i8`
      |
      = note: expected raw pointer `*const u8`
                 found raw pointer `*const i8`
note: function defined here
     --> /Users/user/Documents/Projects/test-esp32c3-app/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-eeec4fa03cb86953/out/bindings.rs:40943:12
      |
40943 |     pub fn esp_tls_conn_new_sync(
      |            ^^^^^^^^^^^^^^^^^^^^^

error[E0308]: mismatched types
   --> /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.49.1/src/private/cstr.rs:168:8
    |
163 |         cstrs[i] = cstr.as_ptr();
    |         ----- here the type of `cstrs` is inferred to be `[*const u8; N]`
...
168 |     Ok(cstrs)
    |     -- ^^^^^ expected `[*const i8; N]`, found `[*const u8; N]`
    |     |
    |     arguments to this enum variant are incorrect
    |
    = note: expected array `[*const i8; N]`
               found array `[*const u8; N]`
help: the type constructed contains `[*const u8; N]` due to the type of the argument passed
   --> /Users/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.49.1/src/private/cstr.rs:168:5
    |
168 |     Ok(cstrs)
    |     ^^^-----^
    |        |
    |        this argument influences the type of `Ok`
note: tuple variant defined here
   --> /Users/user/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/result.rs:532:5
    |
532 |     Ok(#[stable(feature = "rust1", since = "1.0.0")] T),
    |     ^^

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-svc` (lib) due to 4 previous errors
```

SergioGasquez, 2024-12-23:
Hi! The errros are already fixed on git, adding: 
```toml
[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
```
to your Cargo.toml should fix the issues

ivmarkov, 2024-12-23:
> Hi! The errros are already fixed on git, adding:
> 
> ```toml
> [patch.crates-io]
> esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
> esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
> ```
> 
> to your Cargo.toml should fix the issues

Actually you need all of:
```toml
[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
embuild = { git = "https://github.com/esp-rs/embuild" }
```

Alternatively, you can downgrade your nightly to a one from ~ 2 weeks ago, and use that.
Or, just use the `esp` toolchain.

NOTE: We'll have a new release very soon now (as in a week if not a few days).

piemanau, 2024-12-23:
Yes that has worked, thanks guys for the fast replies

closed 2024-12-23 by piemanau