esp-rs/esp-idf-template #257: Template fails on NixOS with 'collect2: ld not found' because `-fuse-ld=mold` is injected into the build process

leonhma, 2025-01-23:
I just generated a project using the template and the cargo option. Building it fails at the final linking step because `ld` (the linker) could not be found. After doing `cargo build -vv` and running the final link command manually with strace, i could see which paths collect2 searches for `ld`. due to `-fuse-ld=mold` being (somehow) added to all relevant cargo commands, it only searches for `ld.mold` in the compiler directories (`.embuild/espressif/...`)- which predictably can't be found. When running the final link command manually without the `-fuse-ld` argument, the linker can proceed, although it then fails due to some strings in the `cargo build -vv` output like "<4 objects omitted>". Do you know where the `-fuse-ld=mold` is set in the build process and how I can unset it?

Here's a repro (including nix flake): https://github.com/leonhma/esprs/tree/6f2da8f16b3646e9498190e5c3835647f4550f2d

ivmarkov, 2025-01-23:
I do not run nixos so I cannot really try to reproduce your problem.

With that said, 
> due to -fuse-ld=mold being (somehow) added to RUSTFLAGS

This is not done by `esp-idf-sys` nor the template. If you are (somehow) doing this yourself, please don't. Ideally, you should figure out what on your os is polluting your RUSTFLAGS and you should remove the offender.

ivmarkov, 2025-01-23:
if this is not clear, `esp-idf-sys` expects a very concrete linker (the espressif flavor of gcc-ld) to be used, and `ldproxy` is there to be exactly able to locate that linker. If you are forcing (via RUSTFLAGS) a different type of linker, there is no way it would work.

leonhma, 2025-01-23:
Turns out the problem was a default in devenv that overrrode the RUSTFLAGS to force mold as the linker. This way the modified ld couldn't be found and so on. So, literally `RUSTFLAGS="-C -fuse-ld=mold"` was the issue 🤦. For anyone else using NixOS and devenv with the same issue, just set `mold.enable` to `false`:
```nix
languages.rust = {
    enable = true;
    channel = "nightly";
    components = [ "rustc" "rust-src" "cargo" "clippy" "rustfmt" "rust-analyzer" ];
    mold.enable = false;
};
```
Personally I think making it the default behaviour to override the default settings of another program is nonsense and i am going to open up an issue over there.

Thanks for your help and confirming this is not an esp-rs/esp-idf issue.

close #257

closed 2025-01-23 by leonhma