esp-rs/esp-idf-template #261: Build failure in esp-idf-sys (v0.36.1) on Windows – "ninja: build stopped: subcommand failed.

berkut0, 2025-03-29:
**Description:**  
I'm encountering a build failure when generating a project via `cargo generate esp-rs/esp-idf-template` and then building it on Windows using Cargo. The build process fails with the following key error:

```
ninja: build stopped: subcommand failed.
```

**What has been done:**  
- Generated a project using the esp-idf-template.
- Attempted to build with Cargo.
- Verified that examples from the "Embedded Rust on Espressif" repository build without issues.

**Key log aspects:**  
- **Long Path Warning:**  
  > Too long output directory: `\\?\D:\dev\rust\std-training\intro\hello-world\target\riscv32imc-esp-espidf\debug\build\esp-idf-sys-0fce1ee41404d522\out`.  
  The warning advises shortening the project path to no more than 10 characters.
  
- **Missing Header Error:**  
  Multiple errors like:  
  > fatal error: bits/c++config.h: No such file or directory  
  This occurs during compilation of C++ components (e.g., `cxx_exception_stubs.cpp` and `cxx_guards.cpp`).

- **Bootloader Build Failure:**  
  The build stops during the bootloader component with an error about failing to open a dependency file.

- **Ninja Error:**  
  The final error consistently reported is:  
  > ninja: build stopped: subcommand failed.

**Assumptions and Observations:**  
- The long output directory paths seem to be causing issues, likely exceeding Windows' path length limitations.  
- I have enabled Windows long path support via policy, but the log remains essentially unchanged, suggesting the template still generates overly nested paths.  
- It appears that the error with "bits/c++config.h" might be indirectly related to the path length issues preventing the proper resolution of include directories.  
- The recurring "ninja: build stopped: subcommand failed." error is common in similar issues and could be a critical pointer to the underlying problem.
- Ihave ESP_IDF_PATH_ISSUES = 'warn' in [env] section of cargo.toml

**Steps to Reproduce:**  
1. Generate a project using:  
   `cargo generate esp-rs/esp-idf-template`  
2. Build the project on Windows with Cargo.

[full_log.txt](https://github.com/user-attachments/files/19521928/log.txt)

Any insights or fixes would be greatly appreciated. Thanks!

ivmarkov, 2025-03-29:
A duplicate of https://github.com/esp-rs/esp-idf-sys/issues/266
* Use <= 10 chars for project path
* Or switch to Linux
* Or switch to Windows WSL

closed 2025-03-29 by ivmarkov

berkut0, 2025-03-31:
@ivmarkov Hi, thanks a lot!
I know it is probably very annoying to explain it again and again, but can you give some insight into why projects from "Embedded Rust on Espressif" compile successfully, but the template does not? Are they not built on template?

ivmarkov, 2025-03-31:
> [@ivmarkov](https://github.com/ivmarkov) Hi, thanks a lot! I know it is probably very annoying to explain it again and again, but can you give some insight into why projects from "Embedded Rust on Espressif" compile successfully, but the template does not? Are they not built on template? (It's important to note that I set the env parameter to warn, so success here is very provisional.)

I don't know. It could be anything - as in, they use slightly shorter project names, or they don't enable certain ESP-IDF functionality, or use a slightly older ESP-IDF which happens to not always use too-long path names in their particular scenaiors, or a combination of these thereof.

In any case, if you are trying to find an "escape hatch" from the long path names issue... such a one simply does not exist. Other than doing the work in https://github.com/esp-rs/esp-idf-sys/issues/266 where a lot of the work might be already done for you or the opposite - you might have to actually upstream a fix or two in GCC proper itself.

berkut0, 2025-04-01:
Thanks for your answer. I'm just going to work through the WSL. It's pretty handy with VS Code and the official extension. The only inconvenience is that you have to keep the whole toolchain in the WSL and actually reinstall it.

It's a bummer that you can't work comfortably under Windows. But apparently there's nothing you can do about that now. Maybe one day Espressif will come to its senses and do something about it.

ivmarkov, 2025-04-01:
Espressif does not really invest in supporting the `esp-idf-*` crates which are a community-driven effort.
Espressif invest paid development work in the bare-metal `esp-hal` crates.

Also two more points:
* The path limitation is less of a problem for the ESP-IDF native SDK (that is, unless you put your C project in a really long project path name) as the native ESP IDF build tends to generate shorter path names, so they don't really have a good reason to strongly push for fixing the issue.
* The GCC-related problems might not even be in the GCC xtensa/riscv toolchains Espressif builds and distributes. It is likely in upstream GCC itself (as I mentioned), so it is not just Espressif who is affected or who is causing the trouble in the first place.

sumukhballal, 2025-06-02:
I have had many projects that have incredibly long names and I used to use hacks to basically get it to work on Windows. I was hoping sometime last year this issue would be resolved at the root but I guess we need a really motivated developer. 

I also used WSL2, but sometimes certain esp features dont work as expected in WSL2 becoming a blocker for some of my projects. Switching to linux was a pain, so I wrote this Rust utility  https://github.com/sumukhballal/cargo-winpath-redirect which has now been tested pretty much locally across about 10 projects with really long names and has been working well for me. If someone wanted to try it, let me know if it helps in the short run!