esp-rs/esp-idf-template #262: CMake template doesn't work with idf 5.3

tobolikd, 2025-03-30:
## Bug description

CMake template doesn't build with idf 5.3 and ends with the following error:

```
error: failed to run custom build command for `esp-idf-sys v0.36.1`

Caused by:
  process didn't exit successfully: `/home/dave/Documents/Github/TOI/sensor/build/esp-idf/rust-sensor/target/release/build/esp-idf-sys-f083c75d9692aa01/build-script-build` (exit status: 1)
  --- stdout
  cargo:rerun-if-changed=/home/dave/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-sys-0.36.1/src/include/esp-idf/bindings.h
  cargo:rustc-env=EMBUILD_GENERATED_BINDINGS_FILE=/home/dave/Documents/Github/TOI/sensor/build/esp-idf/rust-sensor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-b7918bb77462d846/out/bindings.rs

  --- stderr
  /home/dave/Documents/Github/installs/esp-idf/components/freertos/esp_additions/include/freertos/task_snapshot.h:8:2: warning: freertos/task_snapshot.h header is no longer used, and will be removed in future versions. [-W#warnings]
  /home/dave/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-sys-0.36.1/src/include/esp-idf/bindings.h:186:10: fatal error: 'esp_vfs_cdcacm.h' file not found
  clang diag: /home/dave/Documents/Github/installs/esp-idf/components/freertos/esp_additions/include/freertos/task_snapshot.h:8:2: warning: freertos/task_snapshot.h header is no longer used, and will be removed in future versions. [-W#warnings]
  Error: failed to generate bindings in file '/home/dave/Documents/Github/TOI/sensor/build/esp-idf/rust-sensor/target/xtensa-esp32-espidf/release/build/esp-idf-sys-b7918bb77462d846/out/bindings.rs'

  Caused by:
      clang diagnosed error: /home/dave/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-sys-0.36.1/src/include/esp-idf/bindings.h:186:10: fatal error: 'esp_vfs_cdcacm.h' file not found
```

## To Reproduce

From [README-cmake.md](https://github.com/esp-rs/esp-idf-template/blob/master/README-cmake.md)
```
espup install
. $HOME/export-esp.sh
```
```
git clone https://github.com/espressif/esp-idf
git -C esp-idf checkout release/v5.3
esp-idf/install.sh
. esp-idf/export.sh
```
```
cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cmake
cd <your-project-name>
idf.py set-target esp32
idf.py build
```

## Expected behavior

Project builds with idf 5.3 which is in the README-cmake.md

## Environment

- ESP-IDF branch or tag: release/v5.3
- Target device (MCU): esp32
- OS: Fedora 41
- How did you install the environment: described above

## Additional information
- build works with idf 5.2

Vollbrecht, 2025-03-31:
Thanks for reporting! 

I can reproduce the error with the taged v5.3.2 version. Not sure why its complaining currently. They overhauled the vfs component a while ago, and streamlined the usb-cdc/serial-jtag/uart console drivers. And the header itself it is complaining is technically out of the private api, though it still is there in all recent versions.

Vollbrecht, 2025-03-31:
Patching out esp_vfs_cdcacm.h will still throw a followup error, then from "esp_vfs_console.h". And this one is "more serious" because its a normal public api.

Cmake should include it if the "vfs" component is set ( which we do in the template default file ). The cmake logif can be found [here](https://github.com/espressif/esp-idf/blob/a3864c088dafb0b8ce94dba272685b850b46c837/components/esp_vfs_console/CMakeLists.txt#L19) for the esp_vfs_console component ( which is different from the normal vfs component)

ivmarkov, 2025-11-09:
I might have time to play with that by end of next week. Whatever the issue is though, it will only be fixed in the upcoming new release of the esp-idf-* crates.

KingingWang, 2025-11-10:
> I might have time to play with that by end of next week. Whatever the issue is though, it will only be fixed in the upcoming new release of the esp-idf-* crates.

I looked into it; the compilation of the esp-idf-sys crate is failing because two header files are missing. I've opened an issue for esp-idf-sys and hope someone will fix it.

ivmarkov, 2025-11-10:
> I looked into it; the compilation of the esp-idf-sys crate is failing because two header files are missing. I've opened an issue for esp-idf-sys and hope someone will fix it.

Your problem in esp-idf-sys is unrelated to the issue at hand. Folks are complaining here about a linker - rather than a bindgen - issue. And that's with esp idf versions <= 5.3.3