esp-rs/esp-idf-template #263: Build errors in esp-idf-svc when adding ESP IDF components in a mixed C++ / Rust project.

Ant01n3, 2025-04-26:
Hello, thansk for this project!

I have a mixed project C++ / Rust because I would like to use the FastLED library as a component. Everything compiled and worked very well when doing the tests with only FastLED. 

However I wanted to extend the project and use MQTT and the ethernet port on my EPS board. But when adding the necessary components in the CMakeList file, I encounter errors when building the esp-idf-svc crate (no external crate esp_sntp_init, cannot find struct ip_addr_t, etc).

I am quite probably doing a silly error but I cannot find where. What I did : 
- doing a idf.py fullclean,
- doing a idf.py set-target esp32
- running again idf.py menuconfig to enable the FreeRTOS backward compatibility I need for FastLED.
- cleaning any build cache.
- idf.py build -> the process stops when compiling the rust dependencies, at the esp-idf-svc crate.

I also tested creating a blank new project : 
- cargo generate --vcs none --git https://github.com/esp-rs/esp-idf-template cmake
- modifying the components/rust-esp32_test/CMakeList.txt RUST_DEPS : set(RUST_DEPS "pthread" "driver" "vfs" "esp_eth" "esp_netif" "esp_event" "mqtt")
- idf.py set-target esp32
- idf.py build

I have the same errors. There are 77 errors all related to crates or structs or functions not found, I can post a list if needed. 

My ESP IDF version is 5.2 (I tried also 5.1 without success).

Sorry if I missed something in the readme files.

ivmarkov, 2025-04-26:
Perhaps it is best if you paste directly the compiler output.

Ant01n3, 2025-04-27:
Yes sure, here it is, I'm sorry it is quite long : 

`Compiling esp-idf-svc v0.51.0
error[E0432]: unresolved import `esp_sntp_init`
  --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:45:13
   |
45 |     pub use esp_sntp_init as sntp_init;
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^ no external crate `esp_sntp_init`

error[E0432]: unresolved import `esp_sntp_setoperatingmode`
  --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:46:13
   |
46 |     pub use esp_sntp_setoperatingmode as sntp_setoperatingmode;
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no external crate `esp_sntp_setoperatingmode`

error[E0432]: unresolved import `esp_sntp_setservername`
  --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:47:13
   |
47 |     pub use esp_sntp_setservername as sntp_setservername;
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no external crate `esp_sntp_setservername`

error[E0432]: unresolved import `esp_sntp_stop`
  --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:48:13
   |
48 |     pub use esp_sntp_stop as sntp_stop;
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^ no external crate `esp_sntp_stop`

error[E0422]: cannot find struct, variant or union type `ip_addr_t` in this scope
     --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/ping.rs:107:18
      |
107   |         let ta = ip_addr_t {
      |                  ^^^^^^^^^ help: a type alias with a similar name exists: `in_addr_t`
      |
     ::: /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/esp-idf/rust-esp32_ctrl/target/xtensa-esp32-espidf/release/build/esp-idf-sys-2efee025c739147b/out/bindings.rs:17969:1
      |
17969 | pub type in_addr_t = __uint32_t;
      | ------------------ similarly named type alias `in_addr_t` defined here

error[E0422]: cannot find struct, variant or union type `ip_addr__bindgen_ty_1` in this scope
     --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/ping.rs:108:21
      |
108   |             u_addr: ip_addr__bindgen_ty_1 {
      |                     ^^^^^^^^^^^^^^^^^^^^^
      |
     ::: /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/esp-idf/rust-esp32_ctrl/target/xtensa-esp32-espidf/release/build/esp-idf-sys-2efee025c739147b/out/bindings.rs:15702:1
      |
15702 | pub union _ip_addr__bindgen_ty_1 {
      | -------------------------------- similarly named union `_ip_addr__bindgen_ty_1` defined here
      |
help: the leading underscore in `_ip_addr__bindgen_ty_1` marks it as unused, consider renaming it to `ip_addr__bindgen_ty_1`
     -->   /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/esp-idf/rust-esp32_ctrl/target/xtensa-esp32-espidf/release/build/esp-idf-sys-2efee025c739147b/out/bindings.rs:15702:11
      |
15702 | pub union ip_addr__bindgen_ty_1 {
      |           ~~~~~~~~~~~~~~~~~~~~~

error[E0412]: cannot find type `ip4_addr_t` in this scope
     --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/ping.rs:109:32
      |
109   |                 ip4: Newtype::<ip4_addr_t>::from(ip).0,
      |                                ^^^^^^^^^^
      |
     ::: /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/esp-idf/rust-esp32_ctrl/target/xtensa-esp32-espidf/release/build/esp-idf-sys-2efee025c739147b/out/bindings.rs:17969:1
      |
17969 | pub type in_addr_t = __uint32_t;
      | ------------------ similarly named type alias `in_addr_t` defined here
      |
help: a type alias with a similar name exists
      |
109   |                 ip4: Newtype::<in_addr_t>::from(ip).0,
      |                                ~~~~~~~~~
[... 583 lines cut ...]
    |         ^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find function `sntp_set_sync_mode` in this scope
   --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:268:18
    |
268 |         unsafe { sntp_set_sync_mode(sntp_sync_mode_t::from(conf.sync_mode)) };
    |                  ^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find function `sntp_set_time_sync_notification_cb` in this scope
   --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:278:13
    |
278 |             sntp_set_time_sync_notification_cb(Some(Self::sync_cb));
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find function `sntp_get_sync_status` in this scope
   --> /home/antoine/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/sntp.rs:297:35
    |
297 |         SyncStatus::from(unsafe { sntp_get_sync_status() })
    |                                   ^^^^^^^^^^^^^^^^^^^^ not found in this scope

Some errors have detailed explanations: E0412, E0422, E0425, E0432, E0433.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `esp-idf-svc` (lib) due to 70 previous errors
ninja failed with exit code 1, output of the command is in the /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/log/idf_py_stderr_output_1189356 and /home/antoine/Documents/Programs/Research/2025-LEDitbe/imager/rust/esp32_ctrl/build/log/idf_py_stdout_output_1189356
`

ivmarkov, 2025-04-27:
@Ant01n3 Can you try explicitly adding the `lwip` component as well?

Seems to me we have to put even more `cfg`s in `esp-idf-svc` but let's first get a clean build for your use case, and then we can open a bug reports for the additional cfgs.

Ant01n3, 2025-04-29:
Thank you so much @ivmarkov. It seems indeed that `lwip` was the missing component. It compiles now without problem.

ivmarkov, 2025-04-29:
Reopening because we still need to fix esp-idf-svc by adding conditionals that don't enable a bunch of components if `lwip` itself is not enabled.