esp-rs/esp-idf-template #265: Unable to use `esp_idf_svc::ws::client`

anu1667, 2025-06-05:
## Bug description

Unable to use `esp_idf_svc::ws::client`. Using **cargo workspace** for the project. I am reaching out since I couldn't find proper docs around this configuration 

**Setting made so far**
1. Added to crate `Cargo.toml`
```
[package.metadata.esp-idf-sys]
extra_components = [
  { name = "espressif/esp_websocket_client", version = "1.4.0" }
]
```
2. Added to `sdkconfig.defaults` placed in workspace root. And the path is mentioned in `.cargo/config.toml` using `ESP_IDF_SDKCONFIG_DEFAULTS`
```
CONFIG_HTTPD_WS_SUPPORT=y
```

**Error** 
```
error[E0432]: unresolved import `esp_idf_svc::ws::client`
  --> esp32/src/websocket/client.rs:2:22
   |
2  | use esp_idf_svc::ws::client::{
   |                      ^^^^^^ could not find `client` in `ws`
   |
note: found an item that was configured out
  --> $HOME/.cargo/git/checkouts/esp-idf-svc-a28457b0e32c6283/e1d0f28/src/ws.rs:22:9
   |
22 | pub mod client;
   |         ^^^^^^
note: the item is gated here
  --> $HOME/.cargo/git/checkouts/esp-idf-svc-a28457b0e32c6283/e1d0f28/src/ws.rs:13:1
   |
13 | / #[cfg(all(
14 | |     feature = "alloc",
15 | |     esp_idf_comp_tcp_transport_enabled,
16 | |     esp_idf_comp_esp_tls_enabled,
...  |
20 | |     )
21 | | ))]
   | |___^
```

## To Reproduce

use following file structure 
```

├── .vscode
│   ├── settings.json
├── Cargo.lock
├── Cargo.toml             <--------- Outer workspace
├── README.md
├── embedded
│   ├── .cargo                   <--------- build target "xtensa-esp32s3-espidf"
│   ├── .embuild
│   ├── Cargo.lock
│   ├── Cargo.toml           <---------- Inner workspace 
│   ├── README.md
│   ├── build.rs
│   ├── esp32                  <---------- firmware package/crate where ws client is used 
│   ├── sdkconfigs
        ├── sdkconfig.defaults
├── host
│   ├── Cargo.lock
│   ├── Cargo.toml
│   ├── src
├── logic
│   ├── Cargo.lock
│   ├── Cargo.toml
│   ├── src
```

## Expected behavior

The configuration is applied properly and the ws client runs with no errors  

## Environment

- ESP-IDF branch or tag: v5.2.3
- Target device (MCU): esp32s3
- OS: MacOS
- How did you install the environment: espup 0.14.1

Vollbrecht, 2025-06-05:
you may read up on the esp-idf-sys docs regarding the default expected location of the sdkconfig files. Probably in your case they are just ignored. You can set env vars to controll the behaviour for non standard setups as outlined [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp-idf-configuration), E,g sdkconfig are expected to be located relative to the workspace root by default.

anu1667, 2025-06-09:
The configurations mentioned in the doc didn't work for this setup. So we decided to use root package instead of virtual workspace