esp-rs/esp-idf-template #270: ESP32-C6 not compatible with ESP-IDF 5.4.1

espilix, 2025-06-30:
## Error message
```
PS D:\kr> cargo run
   Compiling esp-idf-sys v0.36.1
error: failed to run custom build command for `esp-idf-sys v0.36.1`

Caused by:
  process didn't exit successfully: `D:\kr\target\debug\build\esp-idf-sys-188ecda1a0e12784\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c6",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.4.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
          esp_idf_component_manager: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: RemoteSdk { repo_url: None, git_ref: Tag("v5.4.1") }
  Cloning into 'D:\kr\.embuild\espressif\esp-idf\v5.4.1'...
  fatal: unable to access 'https://github.com/espressif/esp-idf.git/': Could not resolve host: github.com
  Error: Could not install esp-idf

  Caused by:
      command '"git" "clone" "--jobs=36" "--recursive" "--depth" "1" "--shallow-submodules" "--branch" "v5.4.1" "https://github.com/espressif/esp-idf.git" "D:\\kr\\.embuild\\espressif\\esp-idf\\v5.4.1"' exited with non-zero status code 128
```
## Environment

- OS Windows 10 and updated .cargo/config.toml as esp-idf v5.4.1.

espilix, 2025-06-30:
You can resolve issue with this configuration.

```
[build]
target = "riscv32imac-esp-espidf"

[target.riscv32imac-esp-espidf]
linker = "ldproxy"
runner = "espflash flash --monitor"
rustflags = [ "--cfg",  "espidf_time64"]

[unstable]
build-std = ["std", "panic_abort"]

[env]
MCU="esp32c6"
# Note: this variable is not used by the pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.4.1"
IDF_PATH = "C:\\Espressif\\frameworks\\esp-idf-v5.4.1"
ESP_IDF_TOOLS_INSTALL_DIR = "global"
```

closed 2025-06-30 by espilix

ivmarkov, 2025-06-30:
```
fatal: unable to access 'https://github.com/espressif/esp-idf.git/': Could not resolve host: github
```