esp-rs/esp-idf-template #274: Unable to compile on Mac Arm64

adrien3d, 2025-08-05:
Hello, on my M1 MacOS 15.5, I am unable to compile (and therefore run) a vanilla project (just generated with `cargo generate esp-rs/esp-idf-template cargo` and thus, without any modifications).

For an unknown reason, it seems that the libclang.dylib, in esp-clang from the .embuild folder is a x86_64 where I have an arm64 setup.
And I do have a `llvm-esp-16.0.0-20230516-macos.tar.xz` in my `.embuild/espressif/dist`

Did anyone already encounters that issue? I already looked at issues in `esp-generate`, `espup`, `esp-idf-hal` and `esp-idf-sys`, but didn't found a successful solution.

Here is more details of my setup:
```
rustc 1.88.0-nightly (2ab28d2e7 2025-06-24) (1.88.0.0)
binary: rustc
commit-hash: 2ab28d2e728c222edd27f881fd18de24fd88332c
commit-date: 2025-06-24
host: aarch64-apple-darwin
release: 1.88.0-nightly
LLVM version: 19.1.2
```
```
rustup 1.28.2 (e4f3ad6f8 2025-04-28)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.88.0-nightly (2ab28d2e7 2025-06-24) (1.88.0.0)`
```
```           
espup 0.15.1
```

My LIBCLANG_PATH is (after having run . ~/export-esp.sh)
/Users/adrien/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib

And the content of my `export-esp.sh` is: 
```
export LIBCLANG_PATH="/Users/adrien/.rustup/toolchains/esp/xtensa-esp32-elf-clang/esp-19.1.2_20250225/esp-clang/lib"
export PATH="/Users/adrien/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:$PATH"
```

While making a `cargo build`, I am having the following errors:
```
    Compiling compiler_builtins v0.1.158
   Compiling core v0.0.0 (/Users/adrien/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling libc v0.2.172
   Compiling proc-macro2 v1.0.95
   Compiling object v0.36.7
   Compiling unicode-ident v1.0.18
   Compiling std v0.0.0 (/Users/adrien/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
   Compiling libc v0.2.174
  [...]
  
   Compiling embedded-svc v0.28.1
error: failed to run custom build command for `esp-idf-sys v0.36.1`

Caused by:
  process didn't exit successfully: `/Users/adrien/Dev/esp/projects/rust/rustdio/target/debug/build/esp-idf-sys-2944144f2092ab94/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out '24c60e243580c7868f4334a1ba3123481fe1aa48'
  Submodule path 'components/bt/controller/lib_esp32': checked out '5c4a62c1d4577d1352d28708c790ba2b4f741842'
  Submodule path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib': checked out '99ea7d1483a9e5329d0801293d8c051d0fddefd7'
  Submodule path 'components/bt/controller/lib_esp32c3_family': checked out '6470c01165cf4edeed5d826ce4082a90deb92efd'
  Submodule path 'components/bt/controller/lib_esp32c5/esp32c5-bt-lib': checked out '77c67a4d06c95113a2d1766892cde59f5af744af'
[... 448 lines cut ...]
  From ssh://github.com/DaveGamble/cJSON
   * branch            acc76239bee01d8e9c858ae2cab296704e52d916 -> FETCH_HEAD
  From ssh://github.com/espressif/esp-lwip
   * branch            0606eed9d8b98a797514fdf6eabb4daf1c8c8cd9 -> FETCH_HEAD
  From ssh://github.com/espressif/mbedtls
   * branch            98fcfd6d2cea90d306e8fde8e5bffd6087c9cda8 -> FETCH_HEAD
  From ssh://github.com/espressif/esp-mqtt
   * branch            cac1552e62b0474c162547b7cce345d7cd1aecfe -> FETCH_HEAD
  From ssh://github.com/espressif/esp-thread-lib
   * branch            55f18e4cc6a249974247fd408aad79b1049d4b31 -> FETCH_HEAD
  From ssh://github.com/espressif/openthread
   * branch            005c5cefc22aaf0396e4327ee7f2e0ad32a7733b -> FETCH_HEAD
  From ssh://github.com/protobuf-c/protobuf-c
   * branch            abc67a11c6db271bedbb9f58be85d6f4e2ea8389 -> FETCH_HEAD
  From ssh://github.com/pellepl/spiffs
   * branch            0dbb3f71c5f6fae3747a9d935372773762baf852 -> FETCH_HEAD
  From ssh://github.com/ThrowTheSwitch/Unity
   * branch            bf560290f6020737eafaa8b5cbd2177c3956c03f -> FETCH_HEAD
  Using esp-idf v5.3.2 at '/Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/esp-idf/v5.3.2'
  Built components: xtensa, esp_driver_gpio, esp_pm, mbedtls, bootloader, esptool_py, partition_table, esp_app_format, esp_bootloader_format, app_update, esp_partition, efuse, bootloader_support, esp_mm, spi_flash, esp_system, esp_common, esp_rom, hal, log, heap, soc, esp_hw_support, freertos, newlib, pthread, cxx, esp_timer, esp_driver_gptimer, esp_ringbuf, esp_driver_uart, app_trace, esp_event, nvs_flash, esp_driver_pcnt, esp_driver_spi, esp_driver_mcpwm, esp_driver_ana_cmpr, esp_driver_i2s, sdmmc, esp_driver_sdmmc, esp_driver_sdspi, esp_driver_sdio, esp_driver_dac, esp_driver_rmt, esp_driver_tsens, esp_driver_sdm, esp_driver_i2c, esp_driver_ledc, esp_driver_parlio, esp_driver_usb_serial_jtag, driver, esp_phy, esp_vfs_console, vfs, lwip, esp_netif_stack, esp_netif, wpa_supplicant, esp_coex, esp_wifi, bt, unity, cmock, console, http_parser, esp-tls, esp_adc, esp_driver_isp, esp_driver_cam, esp_driver_jpeg, esp_driver_ppa, esp_driver_touch_sens, esp_eth, esp_gdbstub, esp_hid, tcp_transport, esp_http_client, esp_http_server, esp_https_ota, esp_https_server, esp_psram, esp_lcd, protobuf-c, protocomm, esp_local_ctrl, espcoredump, wear_levelling, fatfs, idf_test, ieee802154, json, mqtt, nvs_sec_provider, openthread, perfmon, spiffs, touch_element, ulp, usb, wifi_provisioning, main

  thread 'main' panicked at /Users/adrien/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bindgen-0.71.1/lib.rs:604:27:
  Unable to find libclang: "the `libclang` shared library at /Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/lib/libclang.dylib could not be opened: dlopen(/Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/lib/libclang.dylib, 0x0005): tried: '/Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/lib/libclang.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/lib/libclang.dylib' (no such file), '/Users/adrien/Dev/esp/projects/rust/rustdio/.embuild/espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/lib/libclang.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))"
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

adrien3d, 2025-08-06:
@ivmarkov any hint to try to solve that?

adrien3d, 2025-08-06:
Hum, I am using an arm64 rustc, do you know if I can tell embuild not to use a x86?

Vollbrecht, 2025-08-09:
Ok so to work backwards from the core issue you somehow get a match from the tools.json file in here to [this entry](https://github.com/espressif/esp-idf/blob/v5.3.2/tools/tools.json#L279). We read that tools json file and have our own schema validation. For your case there exist macos and macos-arm64. One layer up we in the embuild crate have a json_schema validation that for this particular case looks like [this](https://github.com/esp-rs/embuild/blob/master/src/espidf/tools_schema.rs#L406). E.g a enum that should cover all cases that can happen in the tools.json.

Than where we actually use that stuff is [here](https://github.com/esp-rs/embuild/blob/master/src/espidf.rs#L251). We try to guess the current host via reading the ENV variables provided by rust. Then we match what we got as good as possible to the schema_json enum we got from esp-idf [here](https://github.com/esp-rs/embuild/blob/master/src/espidf.rs#L297). This code overall is only there to register for the build time the correct PATH for cmake and such that it can find all build tools. Crucially this is **not** installing or downloading the tools itself. That is still done by the standard esp-idf python script that every esp-idf user is using. 

So there are two options. 
 - Either something from the code aboive has a bug in your case, such that we dont correctly export the correct PATH to a libclang version that should have been installed. 
- Something in the esp-idf python tools install is wrong in the host detection. What we do in embuild is collecting what tools need to be installed and than pass it forward to the installer. Here is the relevant [code](https://github.com/esp-rs/embuild/blob/master/src/espidf.rs#L757). E.g if up to that point everything looks fine with respect to what is selected the bug may be in idf_tools.py file. That would be a bug in esp-idf if we dont invoke it wrong.

Other than that i would make sure that you dont set any other env var yourself that is mention in the [esp-idf-sys docu](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#esp_idf_tools_install_dir-esp_idf_tools_install_dir) that somehow could negatively impact this.

I am currently traveling so thats all i can say so far. Hope the info helps you maybe find a clue in your setup.